ShendoXT/memcarduino #6: Commit b4c1d92 not working with MemcardRex 1.8

xadox-1st, 2016-11-28:
After upgrading memcarduino from v2 to b4c1d92 I am not able to read or write memory cards anymore. Is the actual commit not compatible with MemcardRex?

ShendoXT, 2016-11-28:
Communication protocol between PC and Arduino did not change so it's fully compatible with MemcardRex 1.8.
However I did modify the acknowledge detection routine from the Memory Card itself by hooking an interrupt for faster reading instead of predetermined timeout values.

Make sure that Arduino pin 2 is correctly connected to the Memory Card.
What card are you using, 1st or 3rd party?

xadox-1st, 2016-11-29:
I tried an original sony one and a 3rd party one.
[image]
[image]

ShendoXT, 2016-11-29:
I added a testing branch where I introduced a compatible mode which should activate if ACK is not received.
In theory that should make it work for you as it does on v0.2.
I unfortunately don't have Arduino board with me so I can't test it but hopefully it will work. Post results.

You should always use the newest release of Arduino SDK. The reason v0.2 doesn't work is because of
regression in the SDK where functions with 2 byte arguments are not working properly.
My workaround is to use a single uint variable...

TheBlueTroll, 2016-11-30:
Ok, I've just pulled out my arduino uno clone and tried using memcardrex 1.8 to read the latest commit version of the memcarduino project built in arduino ide 1.0.5-r2. it worked fine
[image]
so it may be that you have it wired wrong/ differently, what arduino are you using there, and how do you have your pinout setup in the code?

TheBlueTroll, 2016-11-30:
and just for fun, i also rebuilt and uploaded from 1.6.13 arduino ide the same commit, and it also compiled and worked fine. the memory card in question is a official sony one, i have no third party ones that i could use anyway.

xadox-1st, 2016-11-30:
@TheBlueTroll: I am using a nano clone. it is wired like shown on the picture on my post: https://github.com/ShendoXT/memcarduino/issues/6#issuecomment-263513209

I used the picture by @ShendoXT:
http://2.bp.blogspot.com/-jV7QoFBGHLg/Ucm9LsFZRzI/AAAAAAAAADo/7Brq8LHLKzQ/s1600/MemCARDuino.png

And I have not changed the pinout setup in the code.
Do I have to?

TheBlueTroll, 2016-11-30:
Nope,you don't have to change the pinout, the spi port on the nano is the same as on the arduino, this worked for a third party card, but not a official? are you sure this card works in a ps1/ ps2 properly? the wiring looks fine.
the nano is based upon the ATmega328, which is the same in the arduino uno, only in a smaller package.
what exactly happens that makes you think that memcardrex now doesn't work? does it hang at the progress bar showing no progress, or does it error out detecting?

xadox-1st, 2016-11-30:
Yes, it hang at the progress bar, right at the start.

ShendoXT, 2016-11-30:
I have found out that for some users it just won't work even if everything is set up properly.
It could be the timing, logic levels (maybe that card won't work with 5V logic), or some to me unknown property
messing with the communication. Without having the exact card to try there is not a lot I can do...

I'm curious however if the testing branch will work for you because it will fallback to method used in 0.2
if ack is not received after first command which should increase overall compatibility of the project.

xadox-1st, 2016-12-01:
Yesterday I started from scratch. Installed a fresh Win7 with Arduino IDE 1.6.13.
Uploaded memcarduino 0.4 from the testing branch.

Everything is working great now. Also my original sony memory card is working now.
Tried reading, writing and formating.

I seems also to perform faster than on 0.2.

ShendoXT, 2016-12-01:
Awesome. I'm glad you got it working.

closed 2016-12-01 by ShendoXT