ShendoXT/memcarduino #7: ACK Acknowledge (Pin 2) Realy Needed For Read, Writing or Formatting ?

Refrost, 2017-02-19:
Today I download Lates Version Memcarduino 0.4 
But ı Have Some Problemproblem is 0.4 revision memcarduino is formatting so slowly and if you writing someting memcardrex progress bar is not moving or stuck some time
and if complete writing after ı check my old and new md5's is not correctly
and after ı use hxd hex editor and compare this two item and have some writing errors
after ı see https://github.com/ShendoXT/memcarduino/issues/6 closed post and ı see your post

**ShendoXT  commented on Nov 29 2016 • edited ShendoXT edited this comment 3 months ago** 
" added a testing branch where I introduced a compatible mode which should activate if ACK is not received.
 In theory that should make it work for you as it does on v0.2.
 I unfortunately don't have Arduino board with me so I can't test it but hopefully it will work. Post results.

You should always use the newest release of Arduino IDE. The reason v0.2 doesn't work is because of
 regression in IDE itself where functions with 2 byte arguments are not working properly.
 My workaround is to use a single uint variable to pass parameters."

after ı remove ACK pin and test memory card again 
and now memcard reading writing and formatting is looking so good :D
ı Tested my 5 memory card (2x scph-1020 , 2x scph 102 , 1x cheap ) is now works good

Ack is Realy Needed for this memcarduino software ?

mybe you only check is first and last time acknowledge for writing ?
or mybe check memory card after writing or reading md5 , crc bytes ?

my test spec:
1x arduino nano clone
Memcarduino 0.4 (lastest) , memcarduino 0.3
arduino ide 1.8 , 1.6.3 ,1.6.1

Thanks For This Software For Writing
Best Regards;
Dark Frost

TheBlueTroll, 2017-02-19:
time for me to again, pull out the memcarduino. i have it currenly programmed with mamcardunio 0.3 (from last time), and upon pulling out the ACK pin, the read fails and stalls.
upon updating to 0.4, with the ACK pin in, the read speed is the same as Memcardunio 0.3.
with the ACK pin removed, it also works, albeit a lot slower.

the ACK is really needed for proper use of memory cards because the ACK is part of the PSX memory card specifications. the workaround that was put in at 0.4 was to wait for some time if ACK was not responsive. double check the ACK is properly soldered and whatnot. 
I really cannot say go without ACK because the memcardunio could potentially screw up the card reads and writes without it.

Refrost, 2017-02-19:
Now I Look My Arduino Nano Pins and re soldering all pins and recheck with my multimeter with sound check but ı not have any problem in my Arduino Nano :(
[image]
ı check my memory card port but ı not have any problem :(
[image]
after ı try again my 4 orginal memory cards write slowly (10-15 min) and sometime later all light is open and not write
[image]
if complete write memory card after ı read memory card and save md5 is not equal again
and corrupted save after the reading :(

TheBlueTroll you use arduino nano ?
ı have clone Arduino Nano this is problem or not ?
I dont Know My Problem 0.3 revision slow but good working and 0.4 is not working if remove the ack pins
Maybe is code Code Problem Have but ım not sure

TheBlueTroll, 2017-02-19:
Im using a Arduino Uno, the Nano does work, we know this from issue #6 that it works, and that one was also a clone. 
between 0.3 and 0.4 the only change is handling of the Ack, 0.3 NEEDS it, and 0.4 uses it when it can, else falls back.

im not sure how 0.3 works and then 0.4 breaks, as i have just tested it and its working. so...

What IDE version you using?
are you sure the breadboard is good and not the cause of issues?(i've seen some issues come from rather crap breadboards. try testing from the pins on the nano to the memory card port pins and see if it is less then 1 ohm).
also, how do you feel about python the scripting language?

TheBlueTroll, 2017-02-19:
ive just run 2 tests, 
one with the ACK pin connected.
Total Time:0:00:46.139000
and without
Total Time:0:01:28.789000
it took a full 42 more seconds without it connected, and a md5 of both indicated no read error.

Refrost, 2017-02-19:
I today Resoldering and changing ack ping in my memory card socked 
after ı connected again and tested again
[image]

and not worked again :(
ı try everyting but not worked ı dont know mybe my atmel 328 chip is broken mybe code not compatible arduino nano 

and ı answer your questions
What IDE version you using?
ı tested arduino ide 1.6.13 1.6.1  1.8.1
what your ide version TheBlueTroll ?

are you sure the breadboard is good and not the cause of issues?
ı tested arduino nano
[image]
my little bord and this big board + 5 +3v3 power addon

and now ı wait 30 mins but arduino not writing in my card :(

ShendoXT, 2017-02-26:
Hm. Since you managed to get progress bar going I'll assume your wiring is correct.
Your Memory Card probably doesn't get enough power for successful write operation.
According to specs PS1 power supply gives out ~3.6V. So you really need 3.6V on VCC pin.
Nano's 3.3V output gives out low current so it's not suitable for this. Your external 3.3V power supply might not also cut it.

For a "hack" you can get away with 5V on VCC to have a successful R/W but that should be avoided as a permanent solution.

Refrost, 2017-03-04:
@ShendoXT
I Know your configuration scheme
http://2.bp.blogspot.com/-jV7QoFBGHLg/Ucm9LsFZRzI/AAAAAAAAADo/7Brq8LHLKzQ/s1600/MemCARDuino.png
But if ı not remove ack pin memcarduino not write ı merge two voltage pins one 5v pin but not work
after ı use 3v3 and ı remove ack pin its work again

ı send video please look this
https://youtu.be/cF5bGIfoIlc
(and sorry for darkness and bad quality because my bro is sleeping)

ShendoXT You Use Arduino Nano ?

or how can ı change d2 pin another pin (like d3,d4 or d5)
Because mybe my chip is broken and not send signal

ShendoXT, 2017-03-04:
Yeah, I use Nano but many of the boards have a same processor, ATmega328p.
If you think pin 2 is faulty you can changed the following lines to get ACK on pin 3:
#define AckPin 2 to #define AckPin 3
attachInterrupt(0, ACK, FALLING); to attachInterrupt(1, ACK, FALLING);

Refrost, 2017-03-06:
@ShendoXT 
I Try Your Say But Only AckPin 3 with attachInterrupt(0, ACK, FALLING); Code good Working And Thank You  :)
otherwise I Dont Know But if ı use attachInterrupt(1, ACK, FALLING); ack 3 pin not work :(

closed 2017-09-22 by ShendoXT