ShendoXT/memcarduino #11: A python error occurred when bad checksum.

tosik, 2018-05-14:
I would like to report an error.

https://github.com/ShendoXT/memcarduino/blob/master/memcarduino.py#L195

This line has an undefined variable `res`.

```
Traceback (most recent call last):
  File "memcarduino.py", line 341, in <module>
    memcard_write(f)
  File "memcarduino.py", line 195, in memcard_write
    print "BAD CHECKSUM at frame "+str(i+1)+"/"+str(end)+"  Address:"+ByteToHex(hex_data)+"  CHECKSUM:"+ByteToHex(chk)+" TimeTaken:"+str(tPrint)+str(res)
```

TheBlueTroll, 2018-05-15:
I see, sorry i guess i haven't throughly tested the writing code, but a bad checksum means that the memory card thinks the data sent doesnt match what should have been sent (we dont calculate the checksum,  just write the data) and its up to the game to retry writing the data. However as you can see, the script just ignores it and would continue. When i have time later today i will fix the error (more then likely just remove the str (res) as its not clear what i was using that for) and test it just to be sure.

TheBlueTroll, 2018-05-15:
soo on closer inspection, ive already fixed this bug on my fork, and havent issued a pull request since. soo maby head over to https://github.com/TheBlueTroll/memcarduino while i sort out the pull request.

tosik, 2018-05-16:
Thank you so much!

tosik, 2018-05-16:
Is this line https://github.com/TheBlueTroll/memcarduino/blob/master/memcarduino.py#L287 the fix?  i will try to use your forked version.

TheBlueTroll, 2018-05-17:
sure, if you are still having issues be sure to double and triple check that the connections are correct as a bad checksum on writing is not a normal thing, and it happening usually means the memory card is not getting the data OR that the data is corrupt to begin with.

tosik, 2018-05-18:
I added sleep in each writing loop then no checksum error occurs. Maybe it is my device problem.

closed 2021-04-11 by ShendoXT