ShendoXT/memcarduino #12: Read error: check connections

DieKatzchen, 2018-05-17:
I cannot get this thing to read a card. I'm using an official card, I can put my multimeter into the card slot and read continuity all the way to the correct pin on the chip, and only that pin, and my 3.6v line reads rock stable 3.614v. I don't know what else I can check. It's an elegoo brand nano clone, using a ch340 for the serial chip... could that be it? MemcardRex correctly identifies it as a memcarduino v0.4, though.

TheBlueTroll, 2018-05-17:
well if memcardrex can identify it as v0.4, its on the latest version, and it more then likely is a issue with either elegoo brand nano clone or with how its wired. Double check that all the wires are in the correct pin numbers. also be aware that the card uses a significant amount of power on the 3.6 line

if it continues to not work, you could try using the provided python script to try and test reading the card. it does require downloading python2 and installing pyserial via pip, but i've used it to test reading cards with much more description than memcardrex's read card, and it saves the card allowing you to edit it with memcardrex as well.

DieKatzchen, 2018-05-17:
I did that and I got `error: mc read failure, check connections`. I've tested continuity all the way from the chip to test pads on the memcard, and double checked that I have the correct pins multiple times. I guess the next step is to test it with my genuine Uno and see if that works.

TheBlueTroll, 2018-05-17:
Well mc read failure means that when the arduino tried to read the memory card it got nothing back. How are you generating the 3.6volt for the card? Is the supply dipping when the card starts reading? Whats the voltage when the card is attempted to read?

DieKatzchen, 2018-05-17:
I'm using a 3.6v linear regulator. I thought that might be it too, but when I put my multimeter on it and issued a read request it stayed at 3.614v rock steady.

TheBlueTroll, 2018-05-17:
The only thing left is to assume some type of hardware failure. Setup some led's with current limiting resistors on each of the pins and confirm that they are actually working when trying to read the card, rapid flashing and whatnot. And confirm the memcard works in a real psx.

TheBlueTroll, 2018-05-18:
i've looked up the nano, the pins are the same for the spi interface, and as covered in issue #6 and #7 , arduino nano's and clone nano's have been known to work, but being a cheap clone i would suspect that potentially one of the spi pins (pins 10,11,12,13)may not be functional, and that would result in reading failure.
and again, check that your wiring matches http://2.bp.blogspot.com/-jV7QoFBGHLg/Ucm9LsFZRzI/AAAAAAAAADo/7Brq8LHLKzQ/s1600/MemCARDuino.png
because if it doesnt, it WILL cause this exact error.

pins 10 thru 13 are critical, pin 2 can be optional but recommended, and i have a working reader on my desk right now following those instructions.

DieKatzchen, 2018-05-18:
Wait, I had it wired to chip pins 10-13 and 2. I didn't realize it meant D10-D13 and D2. That explains so much.

TheBlueTroll, 2018-05-18:
Soo its working fine now? Nice. :)

DieKatzchen, 2018-05-19:
Well, I just got off work so i haven't tested it yet, but I suspect that will fix it. Somehow I never realized that the pin definitions are based on functionality and not bare metal pin numbers.

DieKatzchen, 2018-05-19:
Yup. Confirmed working now. Luckily, I wired it so wrong I didn't have to clear the solder out of the holes. Because nothing I had hooked up was supposed to be hooked up at all.

TheBlueTroll, 2018-05-19:
yeah, its fairly common to refer to the pin numbers as found on http://www.pighixxx.com/test/pinouts/boards/nano.pdf not by physical but by the silkscreen. i.e pin 13 refers to D13 and not D9

DieKatzchen, 2018-05-19:
That makes way more sense. This way you don't have to refer to a chart every time you use a different board. The spi pins will always be pins 10-13. The compiler does the work of figuring out what physical pin you're talking about.

closed 2018-05-19 by DieKatzchen