ShendoXT/memcarduino #15: Official playstation 2 memory cardmagic gate 8mb

dimitrisKarampistis, 2019-05-06:
I have followed every instruction on the readme file however it does not seem to working I am using the exe application and it is saying that memory card is not connected. I have checked everything with a multimeter and everything is fine, the card is working I mean the ps2 can access it normally I have tried a testfile .ino and the serial out it just a bunch of F... I have installed the right drivers and I have compiled and the sent the file to and arduino uno, I still can not get it to work

TheBlueTroll, 2019-05-09:
The memcarduino doesnt support ps2 cards due to the magicgate encryption system used on them. It only supports the original playstation's memory card.

frytkisasmaczne, 2020-01-25:
do you think it would work on what i think is a chinese knockoff of a ps2 mc?
[image]

eshicks4, 2020-03-24:
without magicgate the ps2 won't even talk to the card so even knockoffs will need to include it

closed 2021-04-11 by ShendoXT