ShendoXT/memcarduino #22: Gerber PCB for Arduino UNO and NANO

x7l7j8cc, 2021-03-29:
Dont know if someone will be interested, but i designed 2 PCB for both Arduino UNO and NANO.
This is my first time doing this. Hoping will be appreciated.

Note: Any suggestion are Welcome.

Preview (i previewed the PCB in Black Color):
[PSXMemCardUino for UNOr3](https://imgbox.com/XiCBknI0)
[PSXMemCardUino for NANO](https://imgbox.com/xhmVtvxe)

Downloads:
[Gerber_PCB_PSXMemCardUino_2021-03-29.zip](https://github.com/ShendoXT/memcarduino/files/6222297/Gerber_PCB_PSXMemCardUino_2021-03-29.zip)
[Gerber_PCB_PSXMemCardUino_NANO_2021-03-29.zip](https://github.com/ShendoXT/memcarduino/files/6222299/Gerber_PCB_PSXMemCardUino_NANO_2021-03-29.zip)

Pro-Tweaker, 2021-03-30:
Hello,

Thank you very much for this design.

Isn't the MP1584 a step-down regulator ? How can it make 7.5v with 5v input ?

Is it possible to include the BOM (espacially the custom mp1584 board and the switches) ?

Thank you,

PT

Pro-Tweaker, 2021-03-30:
Thank you very much for the extra details.

But for If you read the documentation of the chip it says: 
```
The MP1584 is a high frequency step-down
switching regulator with an integrated internal
high-side high voltage power MOSFET.
```

Also [here](https://www.openimpulse.com/blog/products-page/product-category/mp1584en-mini-dc-dc-step-module/):
```
This DC-DC step down module is based on MP1584EN chip, this is capable of driving a 3 A load and converts input voltage between 4.5 V to 28 V into smaller voltage between 0.8 V to 20 V. 
```

Are you sure about that part ?

Thanks,

PT