ShendoXT/memcarduino #23: PocketStation support

paolo-caroni, 2021-05-02:
With Arduino can be send to the memory card any type of command?
So in theory the PocketStation can be fully supported?

I have a memory card adapter for ps3 but due hardware limitation the PocketStation can't be fully supported on it.
Arduino can be a solution?

TheBlueTroll, 2021-05-02:
in theory, yes. in practice it would take a lot of work,  but i cannot see a hardware reason why the Arduino cannot send any arbitrary command to a pocket station.

the MemCarduino could be extended to support more commands as it does low level communicate on a per byte level (arbitrary byte onto the bus).  but from a look at the no$psx documentation, the pocketstation uses quite a few commands, and might need a full PlayStation emulation running at the same time to correctly communicate for some games.

Kdenk, 2021-08-10:
I don't know what you mean by "fully supported", but I can confirm that I'm able to read/write from/to my PocketStations without problems using MemCARDuino and MemcardRex.

paolo-caroni, 2021-08-16:
> I don't know what you mean by "fully supported", but I can confirm that I'm able to read/write from/to my PocketStations without problems using MemCARDuino and MemcardRex.

Read and write are only two command, the pocketstation have more command, for set date and time for example or for other utility...

I close the issue, because seems that is possible on harware level support all type of command.
Thank you

padraigfl, 2024-01-30:
Just to add to here, I've had issues reading/writing on MemcardRex when the PocketStation has a battery in it, but so far it has worked for me when the battery is removed so if someone is having issues reading/writing maybe try that first.

ShendoXT, 2024-08-05:
Thanks to @padraigfl for kindly donating me a couple of PocketStations.
I have started to implement PocketStation specific commands to MemCARDuino/MemcardRex.

[image]
Also, it looks like PS3 MC Adaptor supports sending custom commands to the interface so it should be possible to do the same on it.

ShendoXT, 2024-08-16:
I updated the MemCARDuino sketch and python 3 script with PocketStation support.

The new commands are:
>--psinfo (print info from pocketstation)
--psbios <output file> (dump bios from pocketstation)
--pstime (appy pc time to pocketstation)

Support is already added to MemcardRex alpha build which I will commit to repository soon.

paolo-caroni, 2024-09-24:
Wow.... I have tryed to add it on the ps3mca card adaptor without luck for years (see ms3mca-ps1 in my repo)... this memcarduino is very very interesting and promising. Also @ShendoXT if you add full support on the ps3mca I will kill myself for incompetence :-)

ShendoXT, 2024-09-24:
Already did 😉
If you compile MemcardRex from repository it comes with PocketStation support for MemCARDuino and PS3 MCA.

And extra flex, I managed to make it fully work on at least one 3rd party adaptor.