ShendoXT/memcarduino #25: Can someone make a video or setup instructions?

Dartomic, 2022-01-20:
Sorry, I know this isn't exactly a project issue. I would have posted this in a disussions section, but your project doesn't appear to have that available.

I know you have setup instructions here, but I don't know any electronics engineering. I did find a video of someone showing how to use this, but he didn't show a voltage divider. And he didn't show what he did to get 7.6v into pin 3. I may have made the voltage divider correctly, but I don't think I did, because it takes several hours to read my memory cards while I have the divider I made on a breadboard connected. I also don't know how to get 7.6v for pin 3. At one point I also ran wires from pins 3, 4, and 5 from the playstation's memory card port while I was trying this with my cards. That might be getting 7.6v, but that was even slower. I really don't know why, but seeing as how it was slower, I can only guess that I probably wasn't even getting 7.6v available at the other end of the pin 3 wire I had plugged into the card reader for the arduino.

I did get really good speeds by using the 5v with one of my knock-off cards, without using the divider to reduce the voltage. This surprised me because I figured I would have fried it. I also fried one of my other cards by doing this, as I would have expected with the one that hasn't fried from me doing that.

I haven't actually soldiered any of the wires. I just have them plugged into a memory card port I ripped out of a broke PSone. That way I can just plug in different memory cards and try them.

I did find a video of someone showing how to use this, but he didn't show a voltage divider. And He didn't show what he did to get 7.6v into pin 3. So I'm not exactly sure what he did. 

Anyway, it would be pretty helpful to see the voltage divider for this project, and how to get that 7.6v, either in a video, or instructions with pictures, for people like me that really don't know anything about electronics engineering. 

Or if someone could show me the divider and kind of explain the placement of the wires and resisters, and the values of the resistance, as well as maybe a good way to get a 7.6v power source value.

TheBlueTroll, 2022-01-21:
Hello @Dartomic
so for the voltage divider, the resistors i use are a 56Ω resistor for R1 and 2 resistors of 113Ω in series to make up the 226 Ω for r2
[image]

[image]

as for the external 7.6. if you are using a official Sony memory card, just leave it unconnected. as inside the memory card it is not connected to anything. if you are using a third party card then the card **MAY** need it, or possibly not. 
unfortunately if your card does need it, then there really is not a easy way to get 7.6v. 
for me i own a cheap variable power supply.
[image]

Dartomic, 2022-01-30:
> Hello @Dartomic so for the voltage divider, the resistors i use are a 56Ω resistor for R1 and 2 resistors of 113Ω in series to make up the 226 Ω for r2 [image]
>
> [image]
>
> as for the external 7.6. if [...]

Oh thanks.  I forgot I asked this, because I was tired when I wrote it. Thanks for taking the time to make that picture with the drawings on it. That definitely looks a lot like the divider I had. Maybe my wires are too thin. Turns out I didn't fry one of my cards though. The data was just too corrupt that wrote to it for memcardrex to read it again. I was able to format it with Unirom and it works fine again. 

Anyway, is it fine to just use the 3.3v pin? I saw a PCB on aliexpress with an arduino soldiered to it to read PS1 cards, and the manufacturer doesn't even bother trying to get power to pin 3. I looked one of my memory card's chips up, made by Atmel. That particular one says it can take 3.3v to 3.6v. So I'm wondering if 3.3v is acceptable for most, maybe all of them. If it's fine to use, then can it be added to the readme as a slower option?

orabeidm, 2022-11-18:
buddy, use [booster cells from aliexpress](https://aliexpress.ru/item/4001066566291.html?spm=a2g2w.orderdetails.0.0.4e724aa6f5kZxD&sku_id=12000027223591155) to get 7.6 volts. you can also get a voltage divider from any of your resistors. to calculate the voltage divider, for example, use the [ELECTRODOC program](https://electrodoc.it/) for phone or any other calculator. take 1/2 ps1 ports, clear it. all this can be connected with wires into a simple case and sealed with a thermal gun ...

[image]
[image]
[image]
[image]
[mycase.zip](https://github.com/ShendoXT/memcarduino/files/10044151/mycase.zip)

[image]

ShendoXT, 2022-11-19:
I like your 3d printed case @orabeidm, very nice 👍

orabeidm, 2022-11-19:
> 👍

Don't use voltage divider. Use 3v on arduino, cause with divider original memcard from JP not working

orabeidm, 2022-11-23:
> I like your 3d printed case @orabeidm, very nice 👍
[image]