ShendoXT/memcarduino #28: lgt8fx nano style support?

orabeidm, 2022-11-20:
[image]
[image]

https://user-images.githubusercontent.com/60073650/202899466-0a7df87c-01af-427a-a3c5-bbaef4691043.mp4

ShendoXT, 2022-11-22:
Looking at the video it looks like you haven't actually written compiled sketch to the device. It says missing bootloader at one point. Try writing a simple hello app which sends something to serial port and verify that it's working properly.

orabeidm, 2022-11-23:
when I record a blink or something like that, everything works. this is strange ... I assembled my copy on another version where I applied 5 volts to the contacts everywhere. but it is interesting what is not so specific with this)

ShendoXT, 2022-11-23:
The way MemcardRex verifies Memcarduino is that it sends GETID (hex 0xA) which should get "MCDINO" ASCII response over serial. After that it asks for firmware version (0xA1). You can fake this simple exchange with a skeleton sketch and it will still verify as genuine Memcarduino. So something in serial communication is not working properly if you don't get anything.

You need to have a working serial connection at 115200 baud.

orabeidm, 2022-11-23:
[image]
It seems to be so ... I don’t know how else to check. I probably need more detailed instructions

ShendoXT, 2022-11-23:
No, that's not what I meant.

Write this simple sketch and verify with serial monitor that it's all working correctly.
You should get "Hello World!" printed each second in the monitor.

void setup() {
2	Serial.begin(38400);
3	}
4	
5	void loop() {
6	Serial.println("Hello World!");
7	delay(1000); 
8	}

orabeidm, 2022-11-23:
I'm sorry, but I don't know what a serial monitor is, how it looks and works.

ShendoXT, 2022-11-23:
https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-serial-monitor

orabeidm, 2022-11-23:
[image]

orabeidm, 2022-11-23:
https://github.com/dbuezas/lgt8fx/issues/21

это должно помочь

ShendoXT, 2022-11-23:
Try this simple skeleton sketch, see if it gets detected as Memcarduino with MemcardRex.

void setup() {
  Serial.begin(38400);
}

void loop() {
  if(Serial.available() > 0){
    switch(Serial.read()){
      default:
        Serial.write(0xE0);
        break;

      case 0xA0:
        Serial.write("MCDINO");
        break;

      case 0xA1:
        Serial.write(0x99);
        break;
      }
    }
}

orabeidm, 2022-11-23:
Compilation error: 'Serial' undeclared (first use in this function)

orabeidm, 2022-11-23:
[image]

ShendoXT, 2022-11-23:
I'm sorry but this looks like it's hardware quirk of some some sort. Maybe bootloader takes more than 2 seconds to run the application in which case detection timeouts.

If you can burn the sketch using ISP without bootloader to start directly or find a way to shorten bootloader wait time.