ShendoXT/memcarduino #29: Can't detect new release with nano v3

carmax91, 2023-01-12:
Hi Shendo,
I've updated my memcarduino with the latest release but memcardrex fails to detect my nano v3 with atmega328p...
If i try to read my mc the program don't detect the memcarduino. Reverted to the old release it works again.

ShendoXT, 2023-01-12:
Hm, try using older MemcardRex with MemCARDuino v0.6 but drop the speed to 38400 manually in the sketch.
I have a feeling that while adding the option to toggle DTR line for Leonardo in newest MemcardRex broke the support for atmega328, sigh...

carmax91, 2023-01-12:
Nothing Shendo. Still Broken!

ShendoXT, 2023-01-13:
Oh, must be MemcardRex then, I'll focus my attention on it to solve this issue. Thanks for the info.
I saw you sent me a PM on discord, if you want I can send you updated MemcardRex so you can test it before I release it.

ShendoXT, 2023-01-16:
New MemcardRex release which fixes this issue is live, closing this issue.

closed 2023-01-16 by ShendoXT