ShendoXT/memcarduino #30: Can't write any data to memory card

DriftLegend, 2023-01-26:
Arduino Nano V3.0 ATMEGA328P CH340
[image]

It can read data just fine, however can't write anything on a memory card, it Always gets stuck at random %.
I use original memory cards and there's definetly no connection issues. Tried using both old and latest software.

[image]

ShendoXT, 2023-01-27:
You are using Arduino with 5V logic. Even though you may be supplying the card with 3.3V the logic is still 5V, and that may damage your Memory Card. You should use a level shifter.
In the past people just connected 5V power supply to the card and got good read/write results with those 5V Arduinos.
But be warned, that may shorten the lifespan of your Memory Card, I'm not recommending it in the long run.

The proper way is to use a MCU with 3.3V logic and use  1K external pullups on Cmd, Data, Att, Clk and Ack lines. I've come to found out in my tests that some cards will not work properly with just using internal pullups.

lf94, 2024-02-05:
Hi, I'm about to attempt to hook this up to an ESP32 - should I be using external pullups also?