ShendoXT/memcarduino #31: Can't communicate with Memcarduino running on Raspberry PI pico

nrayner, 2023-06-21:
I am struggling to get a connection to the raspberry PI pico working. It also doesn't work in Memcardrex on windows. I have no trouble programming the Pico over the serial port in arduino

`nicholas@Macbook ~/D/memcarduino> python3 memcarduino.py --port  /dev/cu.usbmodem14201 -r test

running mcduino check
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
b''
Response Byte is: 

b''
error: mc read failure, check connections

nicholas@Macbook ~/D/memcarduino> python3 memcarduino.py --port  /dev/cu.usbmodem14201 -r test

running mcduino check
error: mcduino communication error, got
b''
" as identifier (should be "MCDINO")
`

nrayner, 2023-06-21:
I have tried this with two different picos and two different memory card ports and got the same result.

I have tested continuity and different memory cards.

I have successfully read memory cards on the same computer when using an esp32.

ShendoXT, 2023-06-21:
Can you verify that the serial communication is working properly on pico? Try running some serial port examples from arduino.

nrayner, 2023-06-22:
Hi

I tried

AnalogInOutSerial
AnalogReadSerial
ReadAnalogVoltage

From the arduino examples and they all worked.

SebastianZientek, 2023-09-10:
Hello,
I had similar issue and I solved it. Maybe it's also your case. I building from Platform IO, and I noticed that `ARDUINO_ARCH_MBED_RP2040` is not set in my environment, so wrong pins are used. I just added `#define ARDUINO_ARCH_MBED_RP2040` before pins definition and it works.

ShendoXT, 2024-04-04:
OK, this is weird.

Apparently when setting up the ack interrupt on Raspberry Pi Pico I have to tell it again that it's a input with pullup.
First time when I was porting the code to Raspberry Pi Pico that was not needed so something changed and it broke MemCARDuino since ack interrupt now was not able to be used.

That would't be such an issue since code should fallback to ack-less mode but it looks like I also didn't give it enough time
in that mode to receive the last byte aka operation status which should return after reading or writing a frame...
It all worked on other MCUs since I used native SPI but for Rpi I'm using it in soft mode because of some SPI issues.

So long story short, one glitch and one oversight and Rpi Pico doesn't work now...

Not to worry though, the fix is simple. Place "pinMode(AckPin, INPUT_PULLUP);" right after "attachInterrupt(digitalPinToInterrupt(AckPin), ACK, RISING);" and it should all work.

I'll make an update ASAP. Sorry for the inconvenience guys.