ShendoXT/memcarduino #36: Not getting serial output on Uno or Nano

Deadking2010, 2024-02-23:
Hoping someone here can help me get this running.

I have tried on both a Uno and Nano clone, uploading the sketch seems fine but I do not get any responses if using the serial monitor other than a bunch of question marks.  I should be able to get some info from the serial monitor, Right??

At this point, I can't even tell if I am accessing the memory card or not.  Maybe someone can point me in the right direction to start.

I have also tried the test branch with the same results.

TheBlueTroll, 2024-02-24:
https://github.com/ShendoXT/memcarduino (Reading saves from a PC)

> To read saves from the Memory Card to your PC use [MemcardRex](https://github.com/ShendoXT/memcardrex/releases) if you are using Windows.
>Make sure to use the latest version because MemCARDuino now runs at 115200bps while older releases used 38400bps.
>For other operating systems you can use a provided Python script.

In short, use [MemcardRex](https://github.com/ShendoXT/memcardrex/releases) to interact with the memcarduino, the serial monitor wont really show anything.

Deadking2010, 2024-02-24:
> https://github.com/ShendoXT/memcarduino (Reading saves from a PC)
>
>
> In short, use [MemcardRex](https://github.com/ShendoXT/memcardrex/releases) to interact with the memcarduino, the serial monitor [...]

Okay, thanks for the input.  I will keep trying.  

Something else I should have asked is if anyone has had luck using Memcardrex under Wine in Linux.  I have set the COM ports to be seen by it but I am not sure if I should be testing with a Windows-based PC instead.  I might try the Python script as well.

TheBlueTroll, 2024-02-24:
I did try using wine to run memcardrex under Linux about 10 years ago(wow time flies) and at the time memcardrex ran, but i couldn't set the serial ports correctly. 
i just tried with wine 9.0 and the latest Memcardrex. attempting to read from the memcarduino causes memcardrex to crash, so i cannot recommend trying to read the card with memcardrex, but the save editing still works perfect.

i would recommend trying via the python script to get the memory card image, then rex to edit it, then the script to write it back.

the python script is a command line only solution to read and write card images (not individual saves).

ShendoXT, 2024-04-05:
Any luck @Deadking2010 in getting this to run?

As @TheBlueTroll already explained serial communication and it's commands are binary and that's why 
you aren't getting anything in the console. You need to interface it either via python script or MemcardRex.

Maybe I should add a simple test command like sending capital T to the console
to get human readable output in the console. Just for cases like this one.

As for linux and wine it should all work, even the serial ports. WinForms are a little ugly under Wine but other
than that should be all good. Make sure to add your user to dialout group for serial access.

TheBlueTroll, 2024-04-05:
> As for linux and wine it should all work, even the serial ports. WinForms are a little ugly under Wine but other than that should be all good. Make sure to add your user to dialout group for serial access.

you are correct, i am a idiot and tried to run memcardrex via WINE (which does crash on serial reads) but should have run it via mono. as its a mono program. and it works perfectly.

ShendoXT, 2024-04-05:
Haha, I even wrote wine and thought mono :)

Deadking2010, 2024-04-05:
> Haha, I even wrote wine and thought mono :)
> 
> New 2.0 will feature dark mode. Now winforms are custom drawn which should move the ugly away. This is alpha build running on winlator which is using mono.
> 
> ![311538510-55c0d945-ca13-42e0-a302-83ea1c18f63f](https://private-user-images.githubusercontent.com/8411572/319991409-3f46a556-00de-43e1-8534-abe3c0d71b41.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIzNDEyMzcsIm5iZiI6MTcxMjM0MDkzNywicGF0aCI6Ii84NDExNTcyLzMxOTk5MTQwOS0zZjQ2YTU1Ni0wMGRlLTQzZTEtODUzNC1hYmUzYzBkNzFiNDEuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MDVUMTgxNTM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzMwYTJlMGNlZjFjNGM3ODk2YjQ0Yjk1MTY1NDk4MzVlNTk3MmI0MjJlNjJmM2IxOWQyMTU3OWUwY2UwMzYzNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.kENcgVcF0cS96SbUapYoVbrVphlrLHW3cB8F78lgCVg)
 
I never did get it working and I kinda gave up on it for the time being as I had other projects clogging up my work space.  I was working with an older laptop and could never get the drivers to install properly.  I will see if I can make some headway on it this weekend as I would love to get mine up and running.