ShendoXT/memcarduino #37: GT2 saves get corrupted when writing them to memory card

AnthonyRyuki, 2024-06-04:
I probably opened an issue describing this situation before, but it turns out that I still can’t write GT2 saves properly. Every time I write these saves, the game will fail to load them, and when I read them back once I wrote them, they do show some differences from the original file when comparing them with an hex editor.

Does it have to do with how big this save file is compared to others? This is in fact the only save file that I can’t copy properly.

I am using an official memory card.

ShendoXT, 2024-07-14:
Can you attach the original save and the read one with differences so I can take a look?

AnthonyRyuki, 2024-07-14:
Well, I just tried doing that again a few times, but no attempt created any differences this time.
But I do have to say, I have reinstalled Windows since I last tried and while I manually set my Arduino to COM1 the first time, after the reinstall it was automatically assigned to COM3 and I just left it alone. I wonder what happened, but I'm happy it's working now. I just hope it stays good... I'll let you know if I have any issues in the future.

ShendoXT, 2024-07-15:
Sure, no worries. Glad to hear that it's working now.