ShendoXT/memcarduino #39: Unable to compile v0.9 with Arduino Nano as target

barrenechea, 2024-09-30:
Hi! Thank you for this awesome project!
I was attempting to update a device I got from AliExpress, and I was unable to compile the project on v0.9 with Arduino Nano as target, getting the following errors:

```
/private/var/folders/7w/dck_kxd55fl0bvyp5fkzv1rh0000gn/T/.arduinoIDE-unsaved2024828-48222-w4ka1e.iq5wn/sketch_sep28a/sketch_sep28a.ino: In function 'byte SoftTransfer(byte)':
/private/var/folders/7w/dck_kxd55fl0bvyp5fkzv1rh0000gn/T/.arduinoIDE-unsaved2024828-48222-w4ka1e.iq5wn/sketch_sep28a/sketch_sep28a.ino:145:22: error: 'CmndPin' was not declared in this scope
         digitalWrite(CmndPin, !!(data & (1 << i)));
                      ^~~~~~~
/private/var/folders/7w/dck_kxd55fl0bvyp5fkzv1rh0000gn/T/.arduinoIDE-unsaved2024828-48222-w4ka1e.iq5wn/sketch_sep28a/sketch_sep28a.ino:148:22: error: 'ClockPin' was not declared in this scope
         digitalWrite(ClockPin, LOW);
                      ^~~~~~~~
/private/var/folders/7w/dck_kxd55fl0bvyp5fkzv1rh0000gn/T/.arduinoIDE-unsaved2024828-48222-w4ka1e.iq5wn/sketch_sep28a/sketch_sep28a.ino:148:22: note: suggested alternative: 'AckPin'
         digitalWrite(ClockPin, LOW);
```

Just to note that I was successfully able to program v0.8 and read a memory card 😄

ShendoXT, 2024-10-01:
Oops, my bad. Can you try it now? Should be fixed with the latest commit.

Also, how are you satisfied with that AliExpress adapter? Are you using 1st or 3rd party memory cards?

barrenechea, 2024-10-01:
Fixed! Thank you very much!

And the adaptor has been working fine so far! It uses the same pinout you've defined, so the stock .ino file works flawlessly. It comes with v0.4 (never used it on that version), and the soldering job on it looks pretty decent. Although the 3D-printed case is not the highest quality (mine does not fully close), it is just a cosmetic thing. I've been testing with a 1st party and with a PicoMemcard 😄

<details>
  <summary>Some pictures for reference</summary>
  
  
[image]
[image]
[image]
[image]

</details>

It came with the new bootloader, so flashing it was pretty straightforward.

closed 2024-10-02 by barrenechea

ShendoXT, 2024-10-02:
Awesome. Glad to see that it's working. Thanks for the pictures also. That adaptor looks neat.
I think I know why they left the controller port also.
Looks like they are repurposing connectors from 3rd party PS2 multitaps.

[image]

Shame that they didn't go with Arduino Micro. It can be used as a HID gamepad.
So the adapter would read Memory Cards with MemcardRex and adapt PS1 and PS2 controllers to PC over USB.
Missed opportunity on their part I must say.

As for pullups, yeah. PS1 has 1K pullups and is able to do 250k clock.
I wanted MemCARDuino to be as straightforward as possible so I used the internal pullups (10K) but had to drop the speed
because otherwise I was getting read/write errors. No big deal, as long as it works :).