ShendoXT/memcarduino #42: Pocketstation and MEGA2560

5a796d, 2024-11-09:
Is the scheme valid for the connection?

I don't want to ruin the pocket station.

I opted for the mega2560 because it has the 3v3 pin

[image]

ShendoXT, 2024-11-09:
It should transfer as the normal save inside a Memory Card image. Unless mega has difficulties communicating with PocketStation (which would be no surprise as PocketStation is difficult to talk to).
What are you using to communicate with MemCARDuino? Python script or MemcardRex?

5a796d, 2024-11-10:
> It should transfer as the normal save inside a Memory Card image. Unless mega has difficulties communicating with PocketStation (which would be no surprise as PocketStation is difficult to talk to). What are you using to communicate with MemCARDuino? Python script or MemcardRex?

memcardrex, at the moment I use an Arduino Uno, Arduino Mega I discarded it because I thought I ran out of Arduino Uno, sorry

I have no problems reading the card but I have doubts as there is no option for pocketstation, I have all the slots full and I am afraid of deleting the saves or formatting the pocketstation (format memory card) from memcardrex. If you can confirm that I can do it easily, I'll proceed

thanks

ShendoXT, 2024-11-16:
If you can read it fine then make a backup of the saves and you are free to try experimenting.
Uno should be able to write to PocketStation normally. Post here if you experience any trouble.